accefa/pren2 #11: Python Projekt entwickeln - Web Schnittstellen

alexander-suter, 2015-02-19:


alexander-suter, 2015-03-01:
Persistierung des JSON Files passiert in dieser Webschnittstelle.

wuethrich44, 2015-03-05:
https://github.com/accefa/rasperry-pi-controller/commit/360d8f32f775398fdb2cb9a9484a01919531834d

alexander-suter, 2015-03-13:
Schnittstelle /image

1. Beim Response sollte noch ein Filename definiert werden. Damit der Browser das Bild direkt darin anzeigt.

2. Warum muss man den Filename übergeben? Der Aufrufer weiss ja gar nicht was das Bild für einen Typ hat. So muss er quasi eine extension angeben. Aber wir arbeiten intern nur mit JPG Bildern.

Sonst sind die Schnittstellen recht gut!
Aber trotzdem :fu:  :smile:

wuethrich44, 2015-03-14:
Der Aufrufer kriegt den Link von GET /camera und ruft image eigentlich nur über den Schlüssel vom JSON auf. :shipit: <- Er hätte das gewusst... :neckbeard: 

Weisst du gerade wie das Header Feld heisst damit sie inline angezeigt werden?

alexander-suter, 2015-03-14:
Ah ok! Alles klar - so passt es natürlich :100:       

P.S. :fu: 

Genau es ist ein header field. warte ...
Content-Disposition: attachment; filename=fname.ext

(http://de.wikipedia.org/wiki/Liste_der_HTTP-Headerfelder)

Aber können wir ja dann am Donnerstag machen. Ich habe die App schon mal installiert und einen Request abgesetzt und das ganze hat gefunkt. Es sind also da nur noch Kleinigkeiten :+1:

alexander-suter, 2015-03-19:
Bild nun als static content mittels web.py ausliefern.

closed 2015-03-19 by alexander-suter