accefa/pren2 #5: Anleitung für GitHub

ninux, 2015-02-19:
Eine kleine Anleitung schreiben wie GitHub zu nutzen ist. Hierzu gehört insbesondere 
* wie die Planung organisiert ist (Milestone, Issue, etc.)
* wie die Struktur der Organisation/Repositories ist

ninux, 2015-02-21:
Ich habe eine erste Fassung erstellt. Kannst du dies bitte kurz durchlesen und auf Korrektheit prüfen?

Das Makefile dazu ist vorbereitet (einfach 'make all' ausführen).

alexander-suter, 2015-02-21:
Hoi Ervin ich habe typos korrigiert. Ich muss nicht sagen, dass die Arbeit sehr gut ist, aber ich denke es ist einfach too much. Aber zu viel ist in der Regel besser als zu wenig. jedenfalls Danke!

Schau doch noch mal das File an.
https://github.com/accefa/pren2/blob/master/tex/github/content/process.tex#L41
Auf Zeile 42 beginnt ein Satz, den ich nicht verstehe. Eventuell umformulieren.

ninux, 2015-02-22:
Vielen Dank für die vielen Sichtungen!

Bezüglich des Umfangs hatte ich auch so meine Zweifel aber ich denke wir können es ja mal so lassen und dann allenfalls kürzen bei der Besprechung im Team am Donnerstag.

Zur Zeile mit dem komischen Satz: Da scheint wirklich was schief gegangen zu sein =) habe mich mal der Sache angenommen.

ninux, 2015-02-26:
Ich habe noch eine Ergänzung gemacht und den Pre-Release erstellt. Falls keine witeren Inputs/Änderungswünsche kommen, dann mache ich den Release daraus. Dannk können wir diese Dokumentation abgeben.

ninux, 2015-02-27:
Per Mail an Dozenten gesendet.

closed 2015-02-27 by ninux